microsoft/AirSim #4965: TF error: Lookup would require extrapolation into the future

nvhungv2k, 2024-07-02:
Hello everyone,
My workspace: Ubuntu 20.04 LTS, ros noetic, Unreal engine 4.27
I ran the [Single drone with monocular and depth cameras, and lidar example](https://microsoft.github.io/AirSim/airsim_tutorial_pkgs/#single-drone-with-monocular-and-depth-cameras-and-lidar) in airsim_tutorial_pkgs
In Rviz, it don't show PointCloud2 from Lidar (), And I checked, it showed:
```
Transform [sender=unknown_publisher]
For frame [drone_1/LidarCustom]: No transform to fixed frame [world_enu]. TF error: [Lookup would require extrapolation 11,472200785s into the future. Requested time 1719913701,674538612 but the latest data is at time 1719913690,202337742, when looking up transform from frame [drone_1/LidarCustom] to frame [world_enu]]
```
I already increased the time of update Lidar by changing update_lidar_every_n_sec in [airsim_node.launch file](https://github.com/microsoft/AirSim/blob/main/ros/src/airsim_ros_pkgs/launch/airsim_node.launch)
But it only delayed the moment of above error.
I think that AirSim/ROS and Unreal aren't same time-synced as ROS and Gazebo.
Any ideal? PLS show me the way how to solve it.

Ikaruga46, 2024-10-20:
Hello there:
Have you solved this?I met the same problem when I used airsim ROS1 pack.But it would not occur when using airsim ROS2 pack ,you can have a try.

JinMohi, 2025-01-08:
Hello，
I am meeting the same problem.My workspace is same as you:Ubuntu 20.04 LTS, ros noetic, Unreal engine 4.27.
Have you solved this?