microsoft/AirSim #4980: MacOs, m2, failed to compile build.bash

554773162, 2024-08-15:
## Question
### What's your question?
./setup.bash has been complied successfully.
But, when I compiled ./build.sh, it reported "

[ 98%] Linking CXX executable ../output/bin/DroneShell
Undefined symbols for architecture x86_64:
  "std::exception_ptr::__from_native_exception_pointer(void*)", referenced from:
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in libAirLib.a(RpcLibClientBase.cpp.o)
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in libAirLib.a(MultirotorRpcLibClient.cpp.o)
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in librpc.a(client.cc.o)
  "___cxa_init_primary_exception", referenced from:
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in libAirLib.a(RpcLibClientBase.cpp.o)
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in libAirLib.a(MultirotorRpcLibClient.cpp.o)
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in librpc.a(client.cc.o)
ld: symbol(s) not found for architecture arm64
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [output/bin/DroneShell] Error 1
make[1]: *** [DroneShell/CMakeFiles/DroneShell.dir/all] Error 2
"

Because the MacOs update, the methods before doesnt work. 
Maybe due to architecture arm64, after modified, it doesnt work either.
I dont know how to deal with it 

#### Context details

hongh0, 2024-08-18:
Same problem, have you figured out it?

Apple clang version 15.0.0 (clang-1500.3.9.4) 
Target: arm64-apple-darwin23.5.0 
Thread model: posix 
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

atapin, 2024-09-09:
@554773162 did you try to `brew install llvm@8`?

shge, 2024-11-07:
Same error on clang 19.1.3.
Tried `brew install llvm@8`, but not found.

```
Warning: No available formula with the name "llvm@8". Did you mean llvm, llvm@18, llm, llvm@17, llvm@16, llvm@15, llvm@14, llvm@13, llvm@12, llvm@11 or wllvm?
```

```
Homebrew clang version 19.1.3
Target: arm64-apple-darwin23.5.0
Thread model: posix
InstalledDir: /opt/homebrew/Cellar/llvm/19.1.3/bin
Configuration file: /opt/homebrew/etc/clang/arm64-apple-darwin23.cfg
```

```
[ 94%] Built target AirLibUnitTests
Undefined symbols for architecture x86_64:
  "std::exception_ptr::__from_native_exception_pointer(void*)", referenced from:
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in libAirLib.a[2](RpcLibClientBase.cpp.o)
  "___cxa_init_primary_exception", referenced from:
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in libAirLib.a[2](RpcLibClientBase.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [output/bin/DroneShell] Error 1
make[1]: *** [DroneShell/CMakeFiles/DroneShell.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Undefined symbols for architecture x86_64:
  "std::exception_ptr::__from_native_exception_pointer(void*)", referenced from:
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in libAirLib.a[2](RpcLibClientBase.cpp.o)
  "___cxa_init_primary_exception", referenced from:
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in libAirLib.a[2](RpcLibClientBase.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [output/bin/HelloSpawnedDrones] Error 1
make[1]: *** [HelloSpawnedDrones/CMakeFiles/HelloSpawnedDrones.dir/all] Error 2
Undefined symbols for architecture x86_64:
  "std::exception_ptr::__from_native_exception_pointer(void*)", referenced from:
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in libAirLib.a[2](RpcLibClientBase.cpp.o)
  "___cxa_init_primary_exception", referenced from:
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in libAirLib.a[2](RpcLibClientBase.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [output/bin/HelloDrone] Error 1
make[1]: *** [HelloDrone/CMakeFiles/HelloDrone.dir/all] Error 2
Undefined symbols for architecture x86_64:
  "std::exception_ptr::__from_native_exception_pointer(void*)", referenced from:
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in libAirLib.a[2](RpcLibClientBase.cpp.o)
  "___cxa_init_primary_exception", referenced from:
      std::__1::promise<clmdep_msgpack::v1::object_handle>::~promise() in libAirLib.a[2](RpcLibClientBase.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [output/bin/HelloCar] Error 1
make[1]: *** [HelloCar/CMakeFiles/HelloCar.dir/all] Error 2
make: *** [all] Error 2
```

arthurhgoldblatt, 2024-11-20:
Any updates on this?