microsoft/AirSim #5001: AirSim + Cesium custom starting location - WARN  [health_and_arming_checks] Preflight: GPS fix too low

Snoeprol, 2024-11-27:
## Question
### What's your question?
I am having issues starting the simulation at a custom location. When I have no OriginPoint, there is no issues.

Now I only suspect one thing, that GPS and the location in the settings file and the location in the engine may be slightly misaligned. Could this really be it? 

Otherwise, does someone else have any suggestions?

But as soon as I add:

```json
    "OriginGeopoint": {
    	"Longitude": 4.902346,
    	"Latitude": 52.379172,
    	"Altitude": 10
    }
```

My drone will not arm and I have the following error:

### Include context on what you are trying to achieve

#### Context details
Ubuntu 22.04
AirSim 1.8.1
PX4 * (HEAD detached at v1.15.0)

```json
{
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "ClockSpeed": 1.0,
    "SubWindows": [
        {"WindowID": 0, "ImageType": 0, "CameraName": "front-left", "Visible": true},
        {"WindowID": 1, "ImageType": 3, "CameraName": "lidar", "Visible": true}
    ],
    "Vehicles": {
        "Drone1": {
            "VehicleType": "PX4Multirotor",
            "EnableGPS": true,
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPortLocal": 14540,
            "ControlPortRemote": 14580,
            "X": 0, "Y": 1, "Z": 0,
            "Cameras": {
                "front-left": {
                    "CaptureSettings": [
                        {
                            "ImageType": 0,
                            "Width": 672,
                            "Height": 376,
                            "FOV_Degrees": 90
                        }
                    ],
                    "X": 0.50, "Y": -0.06, "Z": 0.10,
                    "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
                },
                "lidar": {
                    "CaptureSettings": [
                        {
                            "ImageType": 3,
                            "Width": 640,
                            "Height": 480,
                            "FOV_Degrees": 90
                        }
                    ],
                    "X": 0.50, "Y": -0.06, "Z": 0.10,
                    "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
                }
            }
        }
    },
    "OriginGeopoint": {
    	"Longitude": 4.902346,
    	"Latitude": 52.379172,
    	"Altitude": 10
    }

}

```

```bash
Killing existing PX4 instances...
Starting PX4 instance in /home/mario/PX4-Autopilot/Tools/simulation/../..//build/px4_sitl_default/instance_0...
INFO  [px4] instance: 0

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10015
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 657 bytes, decoded 657 bytes (INT32:16, FLOAT:15)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] PX4_SIM_HOSTNAME: localhost
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator_mavlink] Simulator connected on TCP port 4560.
INFO  [lockstep_scheduler] setting initial absolute time to 1732736162943415 us
WARN  [vehicle_angular_velocity] no gyro selected, using sensor_gyro_fifo:0 1310988
INFO  [commander] LED: open /dev/led0 failed (22)
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
INFO  [uxrce_dds_client] init UDP agent IP:127.0.0.1, port:8888
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2024-11-27/19_36_03.ulg	
INFO  [logger] Opened full log file: ./log/2024-11-27/19_36_03.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
INFO  [mavlink] partner IP: 127.0.0.1
WARN  [health_and_arming_checks] Preflight: GPS fix too low
INFO  [tone_alarm] home set
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: heading estimate not stable
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
INFO  [mavlink] partner IP: 127.0.0.1
WARN  [health_and_arming_checks] Preflight Fail: heading estimate not stable
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [health_and_arming_checks] Preflight Fail: heading estimate not stable
WARN  [health_and_arming_checks] Preflight Fail: Yaw estimate error
WARN  [commander] Arming denied: high throttle	
INFO  [tone_alarm] notify negative
```

### Include details of what you already did to find answers

Checked other answers but to no look. Mostly there's answers for physical drones.

Perhaps this person had the same problem: [link](https://community.cesium.com/t/finding-lat-lon-alt-of-object-not-x-y-z-coordinates/15461/5)

damanikjosh, 2024-12-12:
I also got `Preflight Fail: heading estimate not stable` when setting custom `OriginGeopoint`. I found out that the magnetometer reading is not correct. I fixed this by changing the `getMagField` function of  `AirLib/include/common/EarthUtils.hpp` using [https://github.com/nhz2/XYZgeomag](https://github.com/nhz2/XYZgeomag)

XxM0924, 2025-02-17:
Thank you for your answer

XxM0924, 2025-02-17:
> I also got `Preflight Fail: heading estimate not stable` when setting custom `OriginGeopoint`. Found out that the magnetometer reading is not correct. I fixed this by changing the `getMagField` function of `AirLib/include/common/EarthUtils.hpp` using https://github.com/nhz2/XYZgeomag

Thank you for your answer. He helped me a lot

sidharthmohannair, 2025-03-31:
@damanikjosh @XxM0924 I’m facing the same issue. Can anyone tell me what changes you made in that file? I’m new to this and didn’t understand the repo link you provided.

damanikjosh, 2025-04-16:
I haven't been able to create a fork with the fix, but I can provide you with the modification of EarthUtils.hpp with given library.

https://gist.github.com/damanikjosh/655bd06a5ed0a5454e507dc0022b4ed7