microsoft/AirSim #5012: How to Get an actor's position by tag

Dramano, 2025-03-06:
## Question
### What's your question?

How to get an actor's position by tag?

### Include context on what you are trying to achieve

I'm trying to get an actor's position through Python API `simGetObjectPose` in UE 4.27, it works when I pass the actor's ID Name, but I get `nan` returns when passing the actor's tag.

#### Context details

OS: Windows 11; AirSim Version: 1.8.1; Python Version: 3.8; Unreal version: 4.27.
settings.json:
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ViewMode": "",
  "ClockSpeed": 1,
  "SubWindows": [
    {"WindowID": 0, "CameraID": 0, "ImageType": 0, "Visible": true},
    {"WindowID": 1, "CameraID": 0, "ImageType": 3, "Visible": false},
    {"WindowID": 2, "CameraID": 0, "ImageType": 3, "Visible": true}
    ],
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 3,
        "Width": 100,
        "Height": 80,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      },
      {
        "ImageType": 0,
        "Width": 256,
        "Height": 144,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      }
    ]
  }
}
```
Below is the specific method I used to add tags to the actor and call the `simGetObjectPose` API.

[image]

```python
obj_pose = client.simGetObjectPose("Target")
pos = obj_pose.position  # always get 'nan'
```

### Include details of what you already did to find answers
I tried the solution discussed in #2582, but it didn't work for me.

vikasyadavwork, 2025-04-09:
@Dramano did you solve it ? or still this bug exits ?

Dramano, 2025-04-13:
> [@Dramano](https://github.com/Dramano) did you solve it ? or still this bug exits ?

No, I just use actors' ID Name, it works for me.